MangoAutomation/ma-core-public #502: BACnet Data Source Bugs

terrypacker, 2015-12-09:
1.  If you export the BACnet device configuration via the green down arrow on the discovery tool Mango will hang at shutdown for 600000 ms.

2.  The Device Conversion isn't working correctly, although its unlikely that this can happen:

First install bacnet module
Second delete database
Third restart Mango then setup BACnet data source with points
Fourth stop mango, delete BACnet folder
Fifth copy in BACnet module zip and restart Mango you will get NPE

3. Also there is no label on the down-arrow icon to indicate what it is for.

terrypacker, 2015-12-09:
When this is tested and ready we can upload new version to store.

Puckfist, 2015-12-09:
1. Without using the discovery tool my shutdown time was about 10s. Using the discovery tool then immediately attempting shutdown, my time was about 63s.

2. Was no able to duplicate. Resolved?

3. Down arrow labeled 'Export object list'

Sort of nitpicking, but our JSON export button is labeled 'Export' (opens JSON textbox), our data source CSV button is labeled 'Export CSV' (automatic download), and this arrow is labeled 'Export object list' and downloads an XLSX file. I personally would label anything that downloads a file as 'Download' but I understand the word 'Export' is how we want people thinking about it with the JSON and CSV exports. This one I would really consider a Download, though.

Puckfist, 2015-12-14:
Testing today revealed the WhoIs doesn't seem to be working on the latest. It says its listening, but never finds anything, while 2.2.2 would discover a few devices almost immediately.

terrypacker, 2015-12-17:
Fixed bug where not being able to receive iAm responses would cause the DefaultTransport thread to die and leave the datasource un-responsive to termination and polling.

Fixed bug where adding a data point via the tool BEFORE the data source is saved was displaying an odd popup.

Puckfist, 2015-12-19:
Marking closed. Numerous issues were revealing and fixed in the course of this issue, including:
issue #540
issue #549

And another issue concerning long shutdowns for link-service points, I believe.

closed 2015-12-19 by Puckfist