MangoAutomation/ma-core-public #506: Scripting Data Source - Events from the past

Puckfist, 2015-12-09:
The Scripting data source permits you to specify a timestamp for an outgoing set. If an error occurs on that set, such as a value type error, the event will be raised with the timestamp provided to the set call.

terrypacker, 2015-12-11:
This is now fixed to generate the event at the time the set was made.  It was obviously coded the other way for a reason and it would be good to understand why that was done.

Puckfist, 2015-12-11:
Confirmed. Perhaps it was coded that way for the sake of running a simulation? Or perhaps people's scripts were expected to take a very long time to execute?

...Is there any other way to raise an event at an arbitrary point in time?

closed 2015-12-12 by Puckfist