MangoAutomation/ma-core-public #510: Converting table watchListUsers... Fails on Database Migration

terrypacker, 2015-12-09:
Converting table watchListUsers...
ERROR 2015-12-09 13:18:44,354 (com.serotonin.m2m2.Main.main:120) - Error during initialization
com.serotonin.ShouldNeverHappenException: com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Table 'mango.watchListUsers' doen't exist
        at com.serotonin.m2m2.db.DatabaseProxy.initialize(DatabaseProxy.java:134)
        at com.serotonin.m2m2.Lifecycle.databaseInitialize(Lifecycle.java:588)
        at com.serotonin.m2m2.Lifecycle.initialize(Lifecycle.java:266)
        at com.serotonin.m2m2.Main.main(Main.java:116)
Caused by: com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Table 'mango.watchListUsers' doesn't exist
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:408)
        at com.mysql.jdbc.Util.handleNewInstance(Util.java:411)
        at com.mysql.jdbc.Util.getInstance(Util.java:386)
        at com.mysql.jdbc.SQLError.createSQLException(SQLError.java:1054)
        at com.mysql.jdbc.MysqlIO.checkErrorPacket(MysqlIO.java:4190)
        at com.mysql.jdbc.MysqlIO.checkErrorPacket(MysqlIO.java:4122)
        at com.mysql.jdbc.MysqlIO.sendCommand(MysqlIO.java:2570)
        at com.mysql.jdbc.MysqlIO.sqlQueryDirect(MysqlIO.java:2731)
        at com.mysql.jdbc.ConnectionImpl.execSQL(ConnectionImpl.java:2812)
        at com.mysql.jdbc.ConnectionImpl.execSQL(ConnectionImpl.java:2761)
        at com.mysql.jdbc.StatementImpl.executeQuery(StatementImpl.java:1612)
        at org.apache.commons.dbcp.DelegatingStatement.executeQuery(DelegatingStatement.java:208)
        at org.apache.commons.dbcp.DelegatingStatement.executeQuery(DelegatingStatement.java:208)
        at com.serotonin.m2m2.db.DBConvert.copyTable(DBConvert.java:88)
        at com.serotonin.m2m2.db.DBConvert.execute(DBConvert.java:53)
        at com.serotonin.m2m2.db.DatabaseProxy.initialize(DatabaseProxy.java:131)
        ... 3 more

terrypacker, 2015-12-11:
This was fixed and is ready for testing.  Removed a legacy table from the schema definition in the module.

Puckfist, 2015-12-17:
Looks good.

closed 2015-12-17 by Puckfist