MangoAutomation/ma-core-public #514: Mango JavaScript - Binary & Multistate statistics period times

Puckfist, 2015-12-09:
Assuming periodStart and periodEnd are the names the StartsAndRuntimeList object should have for those properties, for a=binaryPoint.past(DAY);
a["periodStart"] will be null, a.periodStart will be undefined.

The help dialogue lists these properties as realStart and end, but as I mention in [this issue](https://github.com/infiniteautomation/ma-core-public/issues/513) that is misleading.

Puckfist, 2015-12-09:
Also applies to alphanumeric

terrypacker, 2015-12-11:
All statistics objects have the members periodStartTime and periodEndTime so those are now used globally in the help file and toString representations.  I've documented a few new members for the StartsAndRuntime object too.  Please re-check for consistency in all these types.

Puckfist, 2015-12-11:
Check Alphanumeric (ValueChangeCounter) toString() for periodStart*Time* and periodEnd*Time*

Puckfist, 2015-12-18:
Confirmed

closed 2015-12-18 by Puckfist