MangoAutomation/ma-core-public #524: Internal Metrics - Total Work Items

terrypacker, 2015-12-11:
Compare the values on the Internal Metrics page for 'Scheduled work items' vs the value for the data point of 'Scheduled work items'.  The data point is linked to High Priority Work Items.  This was not the case on the jsp page due to how the values were being retrieved from the Monitor.

Also the Scheduled work items is not well understood.  It is actually the current number of HIGH priority work items that are scheduled to run at some point.  This number seems to be ever increasing on some systems and that is not what should be happening.  They should be removed from this count after they are finished, but they must not be getting removed from the RealTimeTimer's queue and this is concerning.

The High priority work items label is also misleading, this is actually the value of concurrently running High Priority items at that instant.  These must fit within the maximum pool size.

Also we should write up some help files for this page.

Puckfist, 2015-12-12:
I see them as being the same on my test, the coop, and a deployed server. I wasn't able to cause them to meaningfully deviate with script shenanigans. Do you have some intuition on how we might get them to deviate?

terrypacker, 2015-12-12:
In the latest code they should be the same.  On the 2.7.0 release code they should be glaringly different.

Puckfist, 2015-12-18:
I haven't been able to observe a disparity, so I am marking closed.

closed 2015-12-18 by Puckfist