MangoAutomation/ma-core-public #526: NoSQL Restore Not Working on Windows

terrypacker, 2015-12-11:
Please test and confirm this is broken.  If so report any stack trace.

Puckfist, 2015-12-11:
Seems like it is working, actually. Earlier I had some points with the same names crossed in checking histories. Points are seemingly having the right count of values to match the origin, but it is interesting to note that purges won't have been applied immediately after performing the backup restore so some points may have much more data than the source system.

I had a discussion with Joel about the nature of purging in the NoSQL database, and it sounds like purge periods are not guaranteed while using NoSQL.

I am not closing this, since there are probably more tests to run on odd usage (i.e. restoring days willy-nilly) but as far as letting it move through the increments the main difference I see is a system with more data (for the points that had been purged on the reference) than its reference.

DVR-Ltd, 2015-12-16:
I don't know if it counts as odd usage, here's what happened:

**Saturday 5th December:**
Hurried decision to switch to a new host made.
Built a new server with Mango 2.6.0 on the new host.
Restored the Mango MySQL database from a recent back up.
Attempted incremental restore of NoSQL data. Process seemed to take appropriate time. Mango contained no data.
Tried to restore again with same result. Probably a couple of times with increasing stress levels.
Our customer told me it would be ok to leave the problem until Monday.

**Monday, 7th December:**
DNS records were updated meaning Mango was now receiving new data (our data comes in by HTTP Listener)
Attempted restoration again, with partial success.
Was able to completely recover data by importing one file at a time. After each import I checked a chart to see that the new data was imported correctly.

**Later:**
Discovered some points in virtual data sources contained historic values which were shown as being set around the time of the first attempted restoration. Manually reset to correct values.

Gary

Puckfist, 2015-12-22:
I'm discovering it's not ideal, but certainly not broken.

One bad behavior is that canceling an incremental restore doesn't really seem to do anything. Meaning, I have 102 files covering 3000000 values over three months. I hit start on the first file and hit cancel as soon as humanly possible. It says 0 seconds, 102 files of 102 files processed. Not running. Navigate to point details, no data, anywhere! Wait a minute, refresh, no data anywhere! Restart Mango, all the data in all the 102 files is recovered.

The second bad behavior seems to be that it doesn't always seem to rebuild its context or ability to query the newly imported values until after a reboot (canceling guarantees), while other times it seems totally fine. This is the case whether you watch the entire restore beginning (seems more likely to have those values displayed without reboot, especially if restored after deleting the MangoTSDB folder as opposed to purging all values) to end or select random files to restore (less likely).

I do believe its working in the sense that if you restore all the data and reboot your Mango you will have all the data, though.

DVR-Ltd, 2015-12-22:
In that case, it's entirely probable that all I had to do was restart Mango. That's good to know. Thanks.

Gary

terrypacker, 2015-12-22:
I would caution doing a restart too soon after a restore.  If there is any recent data in the database Mango will have to queue up data while posting it into the database and this can take some time.  Prior to restarting Mango I would ensure that the 'Point Values waiting to be written' value on the Internal Metrics page has returned to a low number if not 0.

terrypacker, 2016-05-26:
Closing as marked invalid.

closed 2016-05-26 by terrypacker