MangoAutomation/ma-core-public #532: Brewer's Dashboard - missing i18n key

Puckfist, 2015-12-14:
The key for annotation.tankAlarm is not set. Visible on data point details for alarm points, set in MovementDetectorRT and ZoneRateOfChangeDifferenceDetectorRT.

terrypacker, 2015-12-15:
Changing Movement detector to one of: 
```
annotation.alarm.temperatureRising=Temperature rising alarm set
annotation.alarm.temperatureFalling=Temperature falling alarm set
```

Changing Rate of change detector to:
```
annotation.alarm.temperatureZoneDeviation=Temperature zones deviation alarm set
```

Does that make sense?

Puckfist, 2015-12-16:
I would almost be looking for DataSource: {0}
When admin sets, the annotations say: User: admin
Maybe: DS: {0}

Puckfist, 2015-12-18:
Looks like MovementDetector is doing data source ID. Names?

Puckfist, 2015-12-18:
Confirmed

[image]

closed 2015-12-18 by Puckfist