MangoAutomation/ma-core-public #535: Excel Reports - Table view, button permissions

Puckfist, 2015-12-15:
In the Excel Reports module a user will not have the buttons enabled for a report they do not own, even if they have sufficient privileges to view and run the report. They can click the report and gain access to those buttons in the edit div, but they will remain deactivated in the table div. To reproduce, create an excel report and a second user who is also a superadmin.

terrypacker, 2015-12-17:
Fixed by adding permissionProp to the icons in the GridAndEditor.  Note, only users with 'reportEdit' permission will have enabled buttons.

terrypacker, 2015-12-17:
You will need to pickup changes from the CustomDashboards, and Core too see all fixes work.

Puckfist, 2015-12-17:
Tests failed. My non-admin user through an exception when he tried to navigate to the Excel Reports page. I'm not sure if it happened before I had a report I was expecting him to see or not. The stack trace is:

ERROR 2015-12-17 16:14:20,074 (com.infiniteautomation.mango.web.mvc.rest.v1.excelreports.ExcelReportTemplateRestController.queryRQL:114) - AND/OR terms should only have ASTNode arguments 
com.infiniteautomation.mango.db.query.RQLToSQLParseException: AND/OR terms should only have ASTNode arguments
	at com.infiniteautomation.mango.db.query.RQLToSQLSelect.visitAndOr(RQLToSQLSelect.java:227)
	at com.infiniteautomation.mango.db.query.RQLToSQLSelect.visit(RQLToSQLSelect.java:90)
	at com.infiniteautomation.mango.db.query.RQLToSQLSelect.visit(RQLToSQLSelect.java:1)
	at net.jazdw.rql.parser.ASTNode.accept(ASTNode.java:55)
	at com.infiniteautomation.mango.db.query.RQLToSQLSelect.visitAndOr(RQLToSQLSelect.java:220)
	at com.infiniteautomation.mango.db.query.RQLToSQLSelect.visit(RQLToSQLSelect.java:90)
	at com.infiniteautomation.mango.db.query.RQLToSQLSelect.visit(RQLToSQLSelect.java:1)
	at net.jazdw.rql.parser.ASTNode.accept(ASTNode.java:55)
	at com.serotonin.m2m2.db.dao.AbstractBasicDao.createQuery(AbstractBasicDao.java:398)
	at com.serotonin.m2m2.web.mvc.rest.v1.model.PageQueryStream.setupQuery(PageQueryStream.java:42)
	at com.serotonin.m2m2.web.mvc.rest.v1.MangoVoRestController.getPageStream(MangoVoRestController.java:108)
	at com.infiniteautomation.mango.web.mvc.rest.v1.excelreports.ExcelReportTemplateRestController.queryRQL(ExcelReportTemplateRestController.java:112)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:221)
	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:137)
	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:110)
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:806)
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:729)
	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:85)
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:959)
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:893)
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:970)
	at org.springframework.web.servlet.FrameworkServlet.doGet(FrameworkServlet.java:861)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:687)
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:846)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:808)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1669)
	at org.eclipse.jetty.servlets.UserAgentFilter.doFilter(UserAgentFilter.java:83)
	at org.eclipse.jetty.servlets.GzipFilter.doFilter(GzipFilter.java:364)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1652)
	at org.springframework.web.filter.CharacterEncodingFilter.doFilterInternal(CharacterEncodingFilter.java:85)
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:107)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1652)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:316)
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.invoke(FilterSecurityInterceptor.java:126)
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.doFilter(FilterSecurityInterceptor.java:90)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:330)
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:114)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:330)
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:122)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:330)
	at org.springframework.security.web.authentication.AnonymousAuthenticationFilter.doFilter(AnonymousAuthenticationFilter.java:111)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:330)
	at org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter.doFilter(SecurityContextHolderAwareRequestFilter.java:169)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:330)
	at org.springframework.security.web.savedrequest.RequestCacheAwareFilter.doFilter(RequestCacheAwareFilter.java:48)
[... cut ...]
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:97)
	at org.eclipse.jetty.server.Server.handle(Server.java:499)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:310)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:257)
	at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:635)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
	at java.lang.Thread.run(Unknown Source)

terrypacker, 2015-12-18:
Fixed by parsing the user's permissions into separate OR clauses in the RQL Query for the Reports and Templates.

Puckfist, 2015-12-18:
Confirmed

Puckfist, 2016-02-19:
This seems to have broken again.

closed 2016-02-23 by Puckfist