MangoAutomation/ma-core-public #536: Excel Reports - Deleting another user's reports

Puckfist, 2015-12-15:
Very small semi-cosmetic thing:

A superadmin can read the list of finished excel reports. They can select some, assign read permissions, and on ones which they explicitly have read permissions, read reports. The delete button in the report's row stays disabled regardless of read permission. They can use the delete button at the top of the list, though.

I would think superadmin should be able to use those buttons at their leisure. I would also think that behavior of disabled buttons should be correct for everyone but superadmin.

terrypacker, 2015-12-17:
You will need to pickup changes from the CustomDashboards, and Core too see all fixes work.

Puckfist, 2015-12-17:
Mission success!

closed 2015-12-17 by Puckfist