MangoAutomation/ma-core-public #538: Excel Reports - Permission Broken

WoodyIAS, 2015-12-16:
Permission Broken for Excel Reports in Core 2.6.5 build 451

Repro steps

1) Create an Excel Rpt template
2) Login as admin and create an Excel Rpt using the uploaded Excel template
3) Create a Permission Group "Reports"
4) Create a new user and add them to the Reports Group, make sure they are not a "SuperAdmin"
5) Goto the System Setting page and add Reports to the permissions for "Excel report permissions"
Also, edit the Report setup and give group Reports Report read permission.
6) Now logout as admin and login as the user
7) Goto Excel reports and attempt the run the report

Problem: The report is not visible to the user

terrypacker, 2015-12-17:
What you are seeing is NOT a bug, it is the designed behavior.  People with Report Read permissions can only read generated reports. 

From help file:

Report read permission is a list of user groups that define who can view any generated reports.

terrypacker, 2015-12-17:
Closing this as all other permissions problems seem to be contained within the other Git issues for Excel Reports

closed 2015-12-17 by terrypacker