MangoAutomation/ma-core-public #539: CSRF Token - Really old page

Puckfist, 2015-12-17:
I noticed when I have a page that is very old (more than a few minutes), not connected to a running Mango, then I run up the Mango and refresh the page when it's finished booting. I get a normal login page, and I login, but I get:

Expected CSRF token not found. Has your session expired?

Puckfist, 2015-12-17:
Hard to reproduce, I'm finding.

terrypacker, 2015-12-18:
To reproduce, 
1. login to mango and go to data source edit page.
2. terminate Mango
3. start Mango and wait until fully running so you aren't redirected to startup page
4. hit return in browser and you will go to the login page
5. login and see CSRF error page

terrypacker, 2015-12-18:
Test prior to updating your code to ensure you can reproduce it.  If not then comment out the line 58 in RestSecurityConfiguration that was added in the above commit.

Puckfist, 2015-12-19:
Without updating and picking up that line, I am unable to reproduce as you describe. However, reading the code in that commit gave me some clue what may have happened. I may have had it set to launch my browser, which in the course of testing would have given me an abundance of login pages. The changes in that commit only apply to .shtm pages. I am able to produce it every time by opening the login.htm page, killing Mango, restarting Mango, and then using that login page. Strangely I did not get it revisiting unauthorized.htm, but that could be because that does not redirect the browser to login.

terrypacker, 2015-12-21:
I would expect that behavior from the login page.  If you are not refreshing the login page it will still contain the session keys from the previous session.  There is no way for the page to get a new key without hitting the server first.  The security context is mapped to /login.htm but it will only provide session information if that page is requested during the current/active session.

Perhaps an alternative solution here is to provide a better error page for this?

Puckfist, 2015-12-22:
Good solution. Both session timeouts and lingering login pages seem good.

closed 2015-12-22 by Puckfist