MangoAutomation/ma-core-public #540: BACnet - Discovery Tool and Multiple Datasources/Tabs, unpredictable

Puckfist, 2015-12-17:
If someone is engaged in the bizarre practice of having two BACnet data sources listening for and sending out WhoIS from each, sometimes concurrently, they get bizarre behavior. Let's presume Listener B sent their WhoIS first, and is now listening for the results. Their list populated correctly at first, great. Now Listener A comes on and sends out a WhoIS, but alas they get no results! We go check Listener B, and it has double entries for everything. Okay, now cancel Listener A and send out another WhoIS. Listener B's list is suddenly empty! Listener B isn't even listening anything, judging by the state of the buttons but not the red text. 

The results are not exactly the same every time, but it is different shapes of one list getting duplicates, one list getting nothing, and so on. Perhaps it is possible for all datasources to share the discovery device?

terrypacker, 2015-12-17:
I could see this being a problem for more than just the Discovery tool.  Imagine 2 data sources that start at the same time (i.e. at system startup).  Each data source will send out its own WhoIs, the responses will then get clobbered.  I'm not sure how multiple data sources should be handled, probably want to further test this scenario.

Puckfist, 2015-12-17:
WhoIS doesn't seem to work on any port but 47808 - is that coded into the devices at the other end? Seems like its part of the BACnet protocol.
If DS1 is on 0.0.0.0:47808 and DS2 tries to start with local device 192.168.1.x:47808 it will through an address in use exception, start, but never get values.
If DS2 is 192.168.1.x:47809 and DS1 is 0.0.0.0:47808 and they poll the same data point they will both get values. If they are both 0.0.0.0:47808 they will both get values.

Puckfist, 2015-12-17:
From someplace,
Default UDP port:
47808 (X'BAC0')
Configurable:
47808-47832, 49152-65535

So I should probably conclude the BACnet devices visible here are on 47808. I will have to experiment some with change those in the publisher and getting WhoIS

Puckfist, 2015-12-17:
Nevermind, that might only be relevant to BACnet for IPv6.... can we use any other ports than 47808?

Puckfist, 2015-12-17:
Comp1 can have a publisher running on 192.168.1.x:47809 and Comp2 can get its values if everything is properly configured otherwise. Comp2 cannot discover Comp1 with 0.0.0.0:47808, nor with 0.0.0.0:47809. When sending WhoIS messages over 0.0.0.0:47809, wireshark doesn't display any messages with the filter 'bacnet'.

terrypacker, 2015-12-18:
@Puckfist when you say this:
> When sending WhoIS messages over 0.0.0.0:47809, wireshark doesn't display any messages with the filter 'bacnet'.

I believe that is because wireshark probably tied its 'bacnet' filter to port 47808, I would check all network traffic in wireshark without the filter (or filter on the port you are using).

Puckfist, 2015-12-21:
I wonder what would occur if multiple users were trying to discover at the same time?

Also, during testing I noticed the only time I was able to discover my bacnet publisher was when my network was entirely disconnected.

Puckfist, 2018-04-23:
Close as old UI state tracking issue?

terrypacker, 2019-06-19:
Closing as old UI and won't fix

closed 2019-06-19 by terrypacker