MangoAutomation/ma-core-public #542: Excel Reports - Which one am I deleting?!?!

Puckfist, 2015-12-17:
When I use the delete button on the Excel Reports, I get a confirmation popup that asks if I really want to delete the report named XYZ. Well, I do, but there are 7 items in the list with that name. They all have unique XIDs, though. Perhaps the dialogue could ask:

Please press OK to confirm that you want to delete 'name' (XID).

Puckfist, 2015-12-18:
I meant on the Finished / In-Progress Reports portion of the page. The XID isn't visible in the table of possible reports to run! Hopefully the table of possible reports isn't ambiguous. But, on the table of already run reports you can highlight one instance of a report, then click the delete button of a second instance of the report and you'll get a confirmation dialogue with their shared name while the first is colored as selected and the second is on the chopping block.

terrypacker, 2015-12-18:
Adding XID to popup window.  I think that being able to delete the non-highlighted report via the delete icon is acceptable with this change.  

Also fixed another bug where Admin had it's icons disabled for Run-reports that they did not own.

Puckfist, 2015-12-18:
Confirmed. I agree that it's acceptable. It also enabled someone to make deletions while making a shift or ctrl enabled multiple selection.

closed 2015-12-18 by Puckfist