MangoAutomation/ma-core-public #543: Bulk Permissions Update Bypasses Runtime Manager

terrypacker, 2015-12-18:
When setting Bulk permissions they are entered directly into the database as this is more efficient then starting and stopping every data point that has a changed permission.  This issue has a reference to the problem: #353 

The question is: Is this acceptable?  

The second question is: How did @Puckfist access the modbus points 

> I just experienced a situation on a system where the permissions had been set through the bulk update
method, and the modbus data source's points (gotten through runtimeManager by id) was seemingly not informed of a permissions change early the next morning"

terrypacker, 2015-12-18:
Upon review of the code it would be possible to start/stop all necessary data points without restarting the data source.  They are hot-swapped into the running datasource when it is ready.  Still not sure when the permissions are used outside the UI though?

Puckfist, 2015-12-18:
This case was on the Brewer's Dashboard manual control page. Control would have flowed into the setSetpoint method of the tank dsrt.

terrypacker, 2015-12-18:
Fixing to restart a data point if it's permission was changed and it is enabled.  This may open a can of worms but as implemented now it is much more efficient than the alternative of Restarting all necessary data sources.

Things to test:

1.  Setting the permission on a data point that has its enabled boolean set in the DB but has a different data source type than the one it is signed to.  There is some older code that makes a call back to the DB at that point which may be problematic.

2.  Do the changes actually get picked up by all data sources.  If the DS doesn't implement point hot-swapping then it will be a problem.  By default all PollingDataSource(s) have this implemented, its the others we should be concerned about. The same process is used when data points are updated while running so we should be ok.

Puckfist, 2015-12-19:
I can confirm it fixed the original brewer's dashboard case. I will do more testing for those cases...

Puckfist, 2015-12-21:
In the event of (1) it seems nothing horribly bad occurred. I changed the datasourceid of some points right in the database, then used the bulk permissions to update them. I imagine editing the datasource I assigned the points to would have caused streamCorruption, though.

(2) Didn't seem like an issue. The serial data source's points behaved well.

closed 2015-12-21 by Puckfist