MangoAutomation/ma-core-public #548: CSV Import - Chart Colour Validation

Puckfist, 2015-12-18:
Alas there was a side effect of validating the chart colour: Chart Colour isn't exported or imported in a DataPointVO's CSV. This means with a new templateXid all CSV imported points will have a null chart colour and fail to import.

terrypacker, 2015-12-18:
Fixed by updating the model via the template prior to validation of the model.  This probably should have been being done anyway.  This is worth testing thoroughly as we also had an issue in the Core when editing points and changing the template after this change, this specific bug has been fixed.

Puckfist, 2015-12-18:
Not resolved. Consider adding a default chart color next to the PlainRenderer for null templateXid conditions

terrypacker, 2015-12-21:
This is ready for testing again with commit infiniteautomation/ma-modules-private@ecfc6c13a34f47b0477544166f27b1cd5bc86045

Puckfist, 2015-12-21:
Resolved.

closed 2015-12-21 by Puckfist