MangoAutomation/ma-core-public #551: Watchlist - SQL Error on watchlist delete

Puckfist, 2015-12-19:
Steve alerted us to [modules public #13](https://github.com/infiniteautomation/ma-modules-public/issues/13) but I'm moving it here since we're churning through these and I have replication instructions. The error is:

------------
PreparedStatementCallback; SQL [INSERT INTO selectedWatchList (userId, watchListId) VALUES (?,?)]; Unique index or primary key violation: "PRIMARY KEY ON PUBLIC.SELECTEDWATCHLIST(USERID)"; SQL statement:
INSERT INTO selectedWatchList (userId, watchListId) VALUES (?,?) [23505-181]; nested exception is org.h2.jdbc.JdbcSQLException: Unique index or primary key violation: "PRIMARY KEY ON PUBLIC.SELECTEDWATCHLIST(USERID)"; SQL statement:
INSERT INTO selectedWatchList (userId, watchListId) VALUES (?,?) [23505-181]

------------
To replicate, make a few random watchlists with admin1. Now, switch users to admin2 and delete some watchlists' points and then the watchlist.

terrypacker, 2015-12-21:
It was difficult to reliably reproduce but I was able to get it to happen.  Fixed by ensuring that the WatchListDwr.setSelectedWatchList () method wasn't being called 2x when deleting a watch list.

terrypacker, 2015-12-21:
Please test thoroughly as I was not reliably able to reproduce it and so cannot be sure it is fixed.

Puckfist, 2015-12-22:
I was able to reproduce it on old systems at least once per deleting spree, but hopefully never again.

closed 2015-12-22 by Puckfist