MangoAutomation/ma-core-public #555: Permissions - data_point_edit.shtm & data_source_edit.shtm too visible

Puckfist, 2015-12-21:
I can access these two pages while logged in as a user lacking all permissions. I cannot see anything in any lists or edit anything, though.

Puckfist, 2015-12-21:
I found this because New Data Point Details view exposes the edit data point and edit data source buttons even if there is no point selected.

terrypacker, 2016-01-29:
Since we currently don't have a custom permission level for those pages, and they are available to all users I am disabling the buttons on the DataPointDetailsView page if the list is empty.

Puckfist, 2016-01-29:
This is probably fine. I can't seem to do any mischief from there.

It seems from testing the other polishing logic may be disabling those buttons if the user doesn't have data source edit permission

terrypacker, 2016-05-26:
Something to keep in mind for the new UI as permissions should limit the pages one can see.

closed 2017-10-02 by Puckfist