MangoAutomation/ma-core-public #558: Serotonin-JSON special doubles

Puckfist, 2015-12-21:
By setting the discard extreme value ranges to the word Infinity or -Infinity or NaN the special values of the double are used. When this is transported as JSON in the persistent TCP system, these values fail to deserialize, and instead cause JsonException to get thrown.

Puckfist, 2015-12-21:
After a little reading and some test, this looks like it's probably intentional to conform to the JSON spec, where the special values Infinity, -Infinity and NaN are not valid literals to pass. Is that a restriction a Java implementation would want to keep?

Puckfist, 2015-12-22:
This can easily be implemented by adding the following if clauses to the try block at JsonTypeRead:read:80

if("Infinity".equals(element))
	return new JsonNumber(new BigDecimal("123456789"));
if("-Infinity".equals(element))
	return new JsonNumber(new BigDecimal("-123456789"));
if("NaN".equals(element))
	return new JsonNumber(new BigDecimal("987654321123456789"));

But we'd need to decide what's sensible for the values of the BigDecimal, since BigDecimal doesn't like Infinity or NaN. Supporting NaN (will NaN ever be logged if it's discarding extreme values?) seems kluge-y to convert it into a number. Supporting Infinity or -Infinity doesn't trouble me, since we can set a value larger than the largest double and it would be functionally equivalent to positive infinity, since all new values are double before they're BigDecimal.

Puckfist, 2015-12-22:
So, we could do
if("Infinity".equals(element))
return new JsonNumber(new BigDecimal("1.8e309"));
if("-Infinity".equals(element))
return new JsonNumber(new BigDecimal("-1.8e309"));

And not handle the NaN case. You can still set the discard limits to NaN right through the edit data point details section, which is probably an issue. But, this way, you can enter 1.8e309 on the data point details page, save it and see it transform into Infinity, then import/export will still work.

terrypacker, 2016-05-26:
This is rearing its head in the new Audit table when a data point is created/modified.  We will need to look into this and find a good solution.

Puckfist, 2017-04-11:
Did it. https://github.com/infiniteautomation/ma-core-public/commit/2201bc339a71862593366503e6766dbefb2a02be

Puckfist, 2018-06-26:
This is an issue with importing through the new UI, now. I submitted a backup with an Infinity and -Infinity in the serotonin-json, got back

[newUIjsonInfinities.txt](https://github.com/infiniteautomation/ma-core-public/files/2138362/newUIjsonInfinities.txt)

closed 2018-09-11 by Puckfist