MangoAutomation/ma-core-public #559: SNMP Testing Tool - Doesn't return its errors to page

Puckfist, 2015-12-21:
Using the SNMP testing tool, I can get values for a reasonable OID, such as .1.3.6.1.2.1.2.2.1.11.2 as on the Chicken Coop, but .1.3.6.1.2.1.2.2.1.11.2a will present a stack trace of: 

Exception in thread "Thread-68" java.lang.NumberFormatException: For input string: "2a"
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Long.parseLong(Long.java:441)
	at java.lang.Long.parseLong(Long.java:483)
	at org.snmp4j.util.SimpleOIDTextFormat.parse(SimpleOIDTextFormat.java:101)
	at org.snmp4j.smi.OID.parseDottedString(OID.java:141)
	at org.snmp4j.smi.OID.<init>(OID.java:79)
	at com.serotonin.m2m2.snmp.vo.SnmpOidGet.run(SnmpOidGet.java:54)

But it will just say "Getting value..." indefinitely

terrypacker, 2015-12-22:
There is a new version of SNMP4J available at snmp4j.org that we may want to consider using.

Puckfist, 2015-12-22:
Confirmed that you've resolved my issue.

closed 2015-12-22 by Puckfist