MangoAutomation/ma-core-public #561: Global Scripts - Validate Scripts and use Ace Editor

terrypacker, 2015-12-22:
I found that on a test system there was a bad Global Script that was causing ALL scripting functionality to behave oddly (return type was always a function).  The cause was a missing parenthesis in a global script.  

Provide a way to validate the scripts prior to saving them.  

Also add in the Ace editor UI to help find these types of errors before they are saved.

terrypacker, 2016-06-03:
The scripts are validated to the best of our ability.  Added Ace Editor.

Puckfist, 2016-06-03:
Confirmed this seems implemented. I wonder if it is worth adding the little green "run this script" checkmark for consistency and debugging global scripts with print statements. After all, it does have to pass validation to be saved.

terrypacker, 2016-06-03:
I'd say that's not important enough to do right now.

Puckfist, 2016-06-07:
Well, if it becomes worthwhile, it can be achieved with things like...

i18n.properties....
globalScript.scriptValid=Script validated

GlobalScriptDwr....
@DwrPermission(admin = true)
    public ProcessResult validateGlobalScript(GlobalScript gs) {
    	ProcessResult response = new ProcessResult();
    	gs.validate(response);
    	return response;
    }

globalScripts.jsp....
function validateScript() {
    	setUserMessage();
        hideContextualMessages("scriptDetails")
        
        var gs = {
            id: editingScript.id,
            xid: $get("xid"),
            name: $get("name"),
            script: editor.getValue()
        };
        
        GlobalScriptsDwr.validateGlobalScript(gs,
                function(response) {
           if(response.hasMessages)
        	   showDwrMessages(response.messages);
           else
        	   setUserMessage("<fmt:message key="globalScript.scriptValid"/>");
        });
    }

and...
<tr><td class="formLabelRequired">
      			<fmt:message key="common.validate"/> <tag:img png="accept" onclick="validateScript();" title="common.validate"/>
    		</td></tr>

Also, we could consider moving the xid validation from the GlobalScriptsDwr.save function to GlobalScript.validate (where name and the script are validated)

closed 2016-11-07 by Puckfist