MangoAutomation/ma-core-public #563: New Data Point Details - Nonadmin users can't see points

Puckfist, 2015-12-22:
I have a user able to see two data points on the old data point details page, but zero data points on the new one. They're non-admin.

terrypacker, 2015-12-22:
Is it possible that the corresponding data source module doesn't have a point locator model or that model is broken?

terrypacker, 2015-12-22:
The permissions were only being applied after the limit query so in some scenarios the user would end up with 0 points on the page if they didn't have permissions to the points in the limit.

Fixed by applying permissions to all RQL queries from the page.  There is still backing code that will strip out the un-viewable points.  

Ensure we test both default page loads and loads with the point xid in the URL.

terrypacker, 2015-12-22:
Also found that one of the Event Queries was broken when bringing events to the toaster in the top.  Confirm that this is still working on all HTML5 pages when testing.

terrypacker, 2015-12-22:
Also found that someone with read permission could set the value of the point.  This has been changed and should be tested.

terrypacker, 2015-12-22:
New code from data point details module and mango api modules required for testing.

Puckfist, 2015-12-22:
Test failed. It seemed like it wasn't splitting the permissions of points ever at all, and the lowly user could only see a point if one of their groups was the only permission (and no commas!) in either the set or read field.

Puckfist, 2015-12-22:
Passing this test. Bad behavior, displays set box for all points whose locator considers them settable, even if the user does not have permission to set them. Sets return a validation error that is well displayed. I will open an enhancement for hiding the set box...

closed 2015-12-22 by Puckfist