MangoAutomation/ma-core-public #577: Modules Page Upgrade - Cancel Button Not Working

Puckfist, 2015-12-23:
I told a Mango to update, then as fast as humanly possible I clicked the cancel button. The upgrade div disappeared, but some unknown time later my downloads finished and Mango decided to restart / upgrade.

Puckfist, 2016-01-08:
This issue stems from bad labeling in the Core's i18n:
modules.upgradesClose=Cancel
Which seem to make different promises. Upon reviewing the code, absolutely no effort is made to cancel the upgrade. Also, no indication remains on the modules page to indicate the upgrade is still running. Perhaps the Throbber should be moved outside just the upgrades div?

Puckfist, 2016-01-11:
It seems like we probably we want Cancel to work. A volatile boolean checked before each download and the eventual restart? Probably should be done at the same time more logging is inserted. #508

terrypacker, 2016-06-02:
Cancel will stop the downloads and remove the files.

Puckfist, 2016-11-21:
Took me two commits.... https://github.com/infiniteautomation/ma-core-public/commit/9b06ebdc6a755b88e5d5ef770b57605c8a757d4d

closed 2017-02-14 by Puckfist