MangoAutomation/ma-core-public #580: Core - Serial To Server Socket Bridge

Puckfist, 2015-12-23:
Currently there is serial to socket functionality, but this means Mango has to be the outgoing connection. It would be cool, and perhaps useful, if the server side of this was also implemented (accepts incoming connections on some port) this would enable very loopy configurations for test cases.

terrypacker, 2016-05-26:
This has at least partially been implemented.

Puckfist, 2016-05-31:
I may have to ask you to test this one, since I did a fair bit of the implementing.

terrypacker, 2016-11-02:
Tested with Serial Data Source - works ok.  There are some concerns about the client virtual serial port ending up with a dead connection so code was added to the Serial DS to retry connections for writes but not sure that we can do much about incoming messages.

terrypacker, 2016-11-02:
Closing as done for 2.8.0 release.

closed 2016-11-02 by terrypacker