MangoAutomation/ma-core-public #583: Core Security - Add the IP of the client when denied for security

Puckfist, 2015-12-28:
It could be useful to have that information. After all, know thy enemy.

resotek, 2015-12-29:
The login username, if known, should also be included in the error message that is printed to the log file.

Puckfist, 2016-06-03:
Was the intention to only apply this to failed logins? That was the only place I found it. I would have expected from the issue description this to also be relevant to:

1. Non logged in user navigating to .shtm page. Produces message with session cookie at warn level, does not output IP
2. Logged in user navigating to shtm page they do not have permission for, redirects to /unauthorized.shtm, nothing logged at level > DEBUG

terrypacker, 2016-06-03:
Good point, I didn't consider that and will add that code.

Puckfist, 2016-08-15:
I saw it log that a user who isn't logged in navigating to an shtm page produced an IP log, but I did not see one when a logged in user went to a shtm page they did not have access to.

closed 2016-09-12 by Puckfist