MangoAutomation/ma-core-public #585: Jssc - Unchecked null in InputStream

Puckfist, 2015-12-31:
[This forum post](http://infiniteautomation.com/forum/topic/2044/polling-issue-with-high-register-number-with-arduino-via-modbus/10) involves someone getting a whole bunch of NullPointerException thrown from JsscSerialPortInputStream.serialEvent:134

From reviewing this code, it seems to me the only way to generate a null pointer there is the port.readBytes() call on an empty input stream (why are there serial events with an empty buffer? could other threads be parked at the sync and then after the port's bytes are consumed that thread unblocks?), then buffer.length fails. It seems to me that if multiple things can cause failure here (and we're doing the catchall Exception) we may want the stack trace from the attempted read.

Puckfist, 2016-01-14:
I think the solution to his problem was passing the number of bytes to read to the readBytes message from the serial event. Specifically...
this.port.readBytes(); --> this.port.readBytes( event.getEventValue() );
in JsscSerialPortInputStream

Puckfist, 2016-02-09:
I encountered this error on linux as well, using the serial data source to figure out how to talk to a new device. I was unable to get any communication between the two without adding this change.

terrypacker, 2016-02-09:
This change is now in and ready for testing.

Puckfist, 2016-02-10:
Did this make it into seroUtils as well?

terrypacker, 2016-02-11:
It did now, although that shouldn't matter for our Mango code/modules... Just a community service perhaps.

closed 2016-02-11 by Puckfist