MangoAutomation/ma-core-public #586: Graphic Views Shared with all users

Joelhaggar, 2016-01-01:
Even if a graphic view hasn't been shared it shows up for all users.  This seems to only be true for the very first Graphic view in the list but not the others.  Will require more looking into.

terrypacker, 2016-01-29:
Was not able to replicate using this method:
Start with 2 users, admin and catfish.

1. Login as admin and create a view AdminsView (no permissions given to it)
2. Login as catfish (with one permission: user) and go to graphics views, no views available.
3. As catfish create a view CatfishsView
4. Login as admin and go to graphics views, 2 views available.

Puckfist, 2016-01-29:
I was also not able to evince this on the latest core with some trying, either.

My tests included 
1) making many graphical views
2) logging in, viewing s shared view, becoming admin, restricting view access, then trying to view again.
3) removing your own privileges

Puckfist, 2016-02-10:
I am closing, all parties settled on the explanation the observed system had anonymous read access or set access and we were all unable to replicate without using anonymous access.

closed 2016-02-10 by Puckfist