MangoAutomation/ma-core-public #587: Graphic Views are displaying None Level Events on data points

Joelhaggar, 2016-01-01:
Previously None level events would not be displayed on the data points for a graphic view.

Joelhaggar, 2016-01-01:
I think it might display them even after they are acknowledged.

terrypacker, 2016-01-29:
Interesting piece of information here, the NONE level events are acknowledged when they are saved to the database.  They have the acknowledge timestamp set to the active timestamp.  I found a bug where the UserEventsCache was allowing these to be inserted and they would never be removed because they are never acknowledgeable.

Puckfist, 2016-02-02:
Hmm. Something is amiss now....

1) None level events have seemingly stopped showing up entirely. I cannot get them to show up on my events pages querying for None level events. I did verify from the SQL console I had raised immense numbers of None events through a 1ms poll.
2) EventManager:147-148 seems weird from how you describe the acknowledging none (which I found in the EventDao, 80-81). Are None and Do Not Log events which could normally be returned to normal bloating this list? My little bit of investigating certainly suggested there were some Do Not Logs in there of very little value, but it seems like it's doing what it should. Just seemed odd.

Puckfist, 2016-02-02:
From the SQL console I can query for id=X and get my None level event back, but on the legacy alarms page querying for the same id gives me nothing.

Puckfist, 2016-11-11:
This is the case, None level events are visible on the graphical views' points now by design.

closed 2016-11-11 by Puckfist