MangoAutomation/ma-core-public #593: Core Scripting - RuntimeManagerScriptUtility NPE enable/disable

Puckfist, 2016-01-06:
Because the 'getByXid' function is used, the EventDetector list for the point is never populated. This means that when the runtimeManager is asked to save the data point it will throw an NPE when processing it's event detector list.

To fix, call DataPointDao.instance.setEventDetectors(vo); at RuntimeManagerScriptUtility:173 and 199

Puckfist, 2016-01-06:
Adding the method:
@Override
    public DataPointVO getByXid(String xid) {
    	DataPointVO dpvo = super.getByXid(xid);
    	setEventDetectors(dpvo);
    	return dpvo;
    }

to DataPointDao seems like the right solution to me

Puckfist, 2016-01-11:
^ should also have a null check to setEventDetectors or it'll cause problems with the Emporter

terrypacker, 2016-01-14:
Another option would be to just use the DataPointDao.setEventDetectors(DataPointVO) method inside the RuntimeManagerScriptUtility instead of changing the Dao class.  I believe it was done that way in the Dao for performance.

Puckfist, 2016-01-14:
That was my first solution, but I think the change belongs in the Dao, lest the postcondition of getting a correct DataPointVO by its XID be violated (I don't consider it the data point's VO without its event detectors set, do you?)

terrypacker, 2016-03-11:
Add getByXidFull() to the AbstractDao class and implement in necessary classes.

Puckfist, 2016-11-11:
I'm closing this. If we need to implement that getByXidFull method, i'm sure we will when required.

closed 2016-11-11 by Puckfist