MangoAutomation/ma-core-public #594: Core Scripting - RuntimeManagerScriptUtility enables and disables from validation

Puckfist, 2016-01-06:
Calls to the RuntimeManager from the scripting environment are applied immediately regardless of how they got generated. This means that when you're validating your scripts' RuntimeManager functionality it's assertions about enabled or disabling data points are what's actually going on, while your gets and sets depend on whether its a validation run or a real execution to perform their actions. The RuntimeManager should probably do likewise.

terrypacker, 2016-05-02:
This will be accomplished by modifying the ScriptUtils.prepareUtilities method to allow a mock RuntimeManager to be passed in much like the PointValueSetter interface.  We will need to create a RuntimeManagerInterface or abstraction to allow this.

terrypacker, 2017-08-03:
Verified to work as expected test script does not enable a data source.

closed 2017-08-03 by terrypacker