MangoAutomation/ma-core-public #597: Core, Units - kWh not a valid unit

Puckfist, 2016-01-08:
It was observed on the forum that kWh is not a valid unit in the Mango space. Perhaps we should add it as a multiply of joule to the UnitUtil? This is the post: http://infiniteautomation.com/forum/topic/2070/mango-still-not-recognizing-kwh/2

resotek, 2016-01-16:
Although I agree that kWh is not a basic unit, it is a derived unit. I use kWh all the time in the "Integral unit" field of my power readings, which otherwise default to kW·s (not very useful). I type it in as kW\*h, then it is displayed as kW·h. I have also entered kW\*h in the Unit field. It seems to work fine.

Whether or not kWh is added as an aliased unit, I recommend that it be made clear in the unit section of the **context help** that units can be any mathematically valid function by including better examples such as:

* kW\*h
* cm/s
* gal/min

etc.

Puckfist, 2016-01-18:
I agree. I knew enough to know kW/h was really dividing, but I didn't think about that meaning kW*h was probably valid as well. Thanks!

closed 2016-02-11 by Puckfist