MangoAutomation/ma-core-public #601: Graphical Views - sstGraphics not defined on public views

Puckfist, 2016-01-12:
[This commit](https://github.com/infiniteautomation/ma-modules-public/commit/87f92adf399c56820644c5bfb3661f1ee6f66ace) comments out the lines that include the module scripts on public view pages (lines 37 and 38). This is causing an alert when the page is loaded that sstGraphics is not defined, and components which rely on it fail. Was this a purposeful comment out?

Puckfist, 2016-01-15:
[This commit](https://github.com/infiniteautomation/ma-modules-public/commit/86bf531f5c96207ef146a49247176a76b2d3cc49)

Puckfist, 2016-01-29:
This was commented out for the sake of compilation it seems

Puckfist, 2016-01-29:
Seems weird it isn't moduleScripts

Where did the reference to applicationScripts come from?

terrypacker, 2016-01-29:
It must be legacy code as the Common class has moduleScripts and those are what the Module scripts are loaded into at startup.

closed 2016-02-01 by Puckfist