MangoAutomation/ma-core-public #602: Watch List and Graphical Views display Do Not Log as 'unknown alarm level -2'

resotek, 2016-01-16:
Watch List and Graphical Views display Do Not Log as "unknown alarm level -2"

When multiple Do Not Log events occur, multiple "unknown alarm level -2" events are shown, each with its time stamp and alias. Does this mean that Do Not Log events are being logged again?

resotek, 2016-04-21:
I am hoping the screenshot below from Graphical Views would help get this issue resolved. There are many yellow warning rectangles on the graphical views and watch lists that should not be there because the events are configured to "Do Not Log."

[image]

And in the watch list:
[image]

terrypacker, 2016-04-21:
I'm actually working in the area of this code right now. 

1.  Do Not Log events are missing labels for the Graphical View.
2.  Do Not Log events are still fired as normal events, they just do not go into the database.  So you should expect to see them in the UI.
3.  None Level events are logged in the database but not fired so you shouldn't expect to see them in the UI.

Fired events are events that make it into the UI and also get handled by event handlers.

I can see a disconnect here as you are probably looking for an event level that doesn't go into the database AND doesn't get handled.  We don't currently have that...

resotek, 2016-04-21:
Thanks. If Do Not Log events do not go into the database, then where does the Graphical View get the displayed data from?

terrypacker, 2016-04-22:
There is an in-memory cache of recent events that helps limit round trips to the database.  All events except None level events end up in this cache.

resotek, 2016-04-22:
In addition to the "None" events levels, shouldn't the "Do Not Log" events levels also be left out of that cache?

terrypacker, 2016-04-22:
We need to test:

1.  None level events are getting into the userEvents table but not being inserted into the memory cache.  They are also not being sent out on the web socket. (this is wrong). Perhaps we should re-label this to be more descriptive.

2.  Do Not Log level events are going into the cache but not the events or userEvents tables.  

3. Update help files to match whatever the new logic ends up being

terrypacker, 2016-06-02:
Clean up code to be consistent and revisit just prior to release.

Puckfist, 2016-11-03:
Not fully resolved, still showing up on the old Data Point Details page as unknown alarm level -2 under the point value.

Puckfist, 2016-11-03:
I also cannot query for None level events on either events page

terrypacker, 2016-11-04:
Review this after #892

closed 2016-11-07 by Puckfist