MangoAutomation/ma-core-public #605: Add checkboxes on permissions filter search results

resotek, 2016-01-16:
For Mango installations with a lot of points and datasources, it can be complex to get the permissions search results to list only the points the user wishes to change. It would be a great time saver if each row of the search results had a checkbox to select which points to apply the permissions tags to. A checkbox at the top could toggle (invert) all the checkboxes in the search result, just as is done on the watchlist page.

terrypacker, 2016-02-12:
Since we are effectively creating an SQL query with the filters, we will have to verify if any boxes are checked and then build a different type of query such as: ID IN(id1,id2,id3,id4)

terrypacker, 2016-05-25:
Checkboxes have been implemented.

Puckfist, 2016-05-27:
This has not been implemented. I believe there was a misunderstanding here on what the feature request was. There are checkboxes for the filter columns, but thee are not filtering checkboxes on the result table. Possibly these could be worked into the result table, but there are some other limits we could eventually run into, such as the 8192 character limit on the URLs.

terrypacker, 2016-06-02:
Add RQL Input instead with examples.

terrypacker, 2018-03-15:
Closing as this functionality is in the new UI under 'Bulk data point edit'

closed 2018-03-15 by terrypacker