MangoAutomation/ma-core-public #607: Graphical view "type":"IMAGE_CHART" can hog the CPU

resotek, 2016-01-16:
**Problem:** When displaying graphical views with IMAGE_CHART, the CPU utilization climbs significantly.

**Likely cause:** The graphical view object "type":"IMAGE_CHART" generates a png chart each time there is a context update. If the user is plotting 1-minute data on a 1-day time span for five datapoints, each chart will contain 1440*5=7200 datapoints. If the chart were only generated once per minute (the logging rate of the datapoints) that would not be so bad. However, the chart is generated whenever there is a context update. So if the points are polled once per second, even though they are only logged once per minute, there will be 5 charts generated every second.

Solution: Add a time interval field to the IMAGE_CHART configuration dialog, so that the user can configure the minimum update interval. If left blank, update the chart on context update; otherwise, update the chart only when the indicated time interval has elapsed.

I consider this a serious bug, as it should not be so easy to hog the CPU. I have many chart objects that plot short interval (1 minute or 1 second) data, and each time an additional user views those pages the CPU idle time goes way down.

terrypacker, 2016-02-12:
This feature is being implemented for all view components but currently will only be available to the image chart component.  Would this be useful for any other components?

Puckfist, 2016-02-17:
Everything looks good. Only blemish I found was that the JSON import doesn't validate the period type, so through the importer I could do MILLSECONDS, MONTHS, or YEARS and it will successfully import. The component will display 'seconds' while being edited, and saving the component will set it to 'seconds' again. Fix by either enabling those time periods in the drop down or prevent the import, methinks.

terrypacker, 2016-02-17:
I've added JSON Validation.

closed 2016-02-17 by Puckfist