MangoAutomation/ma-core-public #608: Split the Event Handler page into a right/left pane like the Watch List page

resotek, 2016-01-16:
**Problem:** Currently the Point event detectors listed individually. That list can be very long, requiring a lot of scrolling. Each time another event or event handler is selected, the user must scroll all the way up to the top of the page to view or edit the event handler. Then, to select an adjacent event, the user must once again scroll all the way back to where they were. This is problematic when there are many events. 
The Event handler event edit should not be at top  of page. Event list should be in left panel like watchlist, and event handler edit box should be in right panel due to excessive vertical scrolling when there are many events.

**Solution 1:** Split the event page into a right and left section so that the Event Types in the left section can be scrolled without the Event Handler section being scrolled too. 

**Solution 2:** Group Point Event Detectors by Device Name or by Datasource in the Event types tree list.

I recommend implementing both solutions.

terrypacker, 2016-02-12:
When we redo the UI this will be one of the goals.

terrypacker, 2018-03-15:
Closed and opened infiniteautomation/ma-dashboards#137

closed 2018-03-15 by terrypacker