MangoAutomation/ma-core-public #609: Add small script box to event handler

resotek, 2016-01-16:
Add asmall script to event handler, like the one on the point link page, so you can invert, divide, etc.

terrypacker, 2016-05-02:
This will be a new type of Event Handler and is in the 2.8.x release stream for a later 2.8.x release.

Puckfist, 2017-04-18:
I have added this.

There may remain some completion, 1) No ScriptPermissions set (it has an empty ScriptPermissions). 2) No scriptlog for set point handler RT. Other scripts have scriptlogs, this one doesn't yet!

And there may be some improvement too, like 1) documentation. How to not set a value? 2) how to wrap the event object? Currently only the target point is provided to the context, but things can be accessed through the DataPointQuery.query but not without script permissions!

closed 2017-08-03 by Puckfist