MangoAutomation/ma-core-public #610: User defined sstgraphics objects are lost during upgrades

resotek, 2016-01-16:
**Problem: ** I have many sstGraphics objects that must be updated manually each time Mango is upgraded. This is a tedious and sometimes risky task, because if this is not done at the right moment during the installation, recovering those lost object requires them to be reinstantiated on each graphical view page where they were used.

**Solution:** add support for placing user-supplied graphic objects in the overrides folder.

resotek, 2016-01-20:
Since Mango restarts cause long data outages and obviously disable point links, it would be very useful for there to be a 'reload graphics' function that can reload new graphics without requiring a full Mango restart.

Puckfist, 2016-11-07:
This title was solved by issue #359 but the second issue you raise remains, so I will retitle this issue and use it for that.

Puckfist, 2018-03-19:
It looks like being able to call this function would require modifications suiting a minor release. If we are going to do this one (which is debatable since this graphics apparatus is not the encouraged dashboarding technique any longer), it would likely require using a different kind of concurrent list (CopyOnWriteArrayList?) for the image set and changing the signature for the loading method to be public and accessible someways