MangoAutomation/ma-core-public #611: ma.log waiting for low priority thread should report number of threads left

resotek, 2016-01-16:
**Problem:** Sometimes during shutdown the ma.log reports that mango is waiting for low priority threads to complete. This can take a while. The problem is that there is no indication as to how much progress, if any, is being made to that end.

**Solution**: ma.log should log the number of threads left to complete, so the user can determine whether to wait or kill the process.

terrypacker, 2016-05-26:
There is only 1 low priority thread in Mango but it has an unbounded queue of tasks it must process.  New logging messages have been added to show the size of the queue.  Mango should never take more than 6 minutes to terminate all 3 of the high, medium and low priority tasks.  

This could be made adjustable...

terrypacker, 2016-06-02:
Make the time to shutdown configurable via env.properties.

terrypacker, 2016-06-03:
Perhaps we want a shorter default shutdown time than 6 min + 6 min = 12 minutes?

Puckfist, 2016-07-06:
The comments in the env.properties file do not reflect the reality. It says that each tick is 10 seconds, but it is clearly 5 in BackgroundProcessing lines 381 && 383. I suppose that's 10 seconds if you have tasks in both categories to wait for? Mine was only waiting 5 when I breakpoint'ed an email handler on shutdown.

Also, the log statement in Lifecycle:469 says waiting for high and low priority tasks, but it's really only waiting on the high priority tasks, right?

Puckfist, 2016-08-15:
Seems like the log messages would still be misleading, since it says " waiting " + rewaits + " more seconds" but it is possible to sleep for 2 seconds each decrement of rewaits if low priority tasks also need termination.

terrypacker, 2016-11-09:
Tested and working as advertised.

closed 2016-11-09 by terrypacker