MangoAutomation/ma-core-public #613: Allow CSV import/export of filtered Data Point Permissions

resotek, 2016-01-16:
Allow CSV import/export of Data point permissions for external editing or external script-based generation.

terrypacker, 2016-05-26:
This doesn't fit with the table and how it performs a bulk update with a WHERE clause, but we could still provide a place to use the query to download a file.  When uploading a file the results would be be displayed in the table unfortunately.  This probably means it won't fit on this page or if it does the results of the upload will be limited to a count of points edit.

terrypacker, 2016-06-02:
Create an RQL input box on this page that you can use as an alternative/in addition to the filter boxes.

terrypacker, 2016-06-02:
Be sure to add examples of useful queries.

Puckfist, 2018-04-17:
Perhaps putting an export on the new bulk data point edit page to export to either CSV or JSON would be the closure?

terrypacker, 2019-06-19:
Closing as this can be done in the UI now on the Bulk Data Point Edit page

closed 2019-06-19 by terrypacker