MangoAutomation/ma-core-public #614: Core - Copy point doesn't copy DataPointVO

Puckfist, 2016-01-18:
The 'Copy point' button that is built into the row for every data point on a data source does not copy the rest of the DataPointVO's settings. This makes copy less fruitful than it could be!

Puckfist, 2016-01-19:
This is caused by the Template. The template is applied even if the source point for the copy didn't have a template enabled.

terrypacker, 2016-02-11:
I took a crack at this but it will take some thought.  We currently don't have a way to track if a data point is a copy and all that dwr code is quite verbose.  So by the time the copy is saved we fall into the default logic for a new data point.  I think the same thing is happening with the event detectors as they get copied during a full data source copy but not a single point copy.

terrypacker, 2016-02-11:
Consider using the concept of a clipboard where we push on a data point and then pull it off during the copy.

Puckfist, 2018-04-06:
Solution was not applying the default template for copied points, instead use source point's template

closed 2018-04-06 by terrypacker