MangoAutomation/ma-core-public #615: Display values and timestamp for metadata sources, and return value and timestamp

resotek, 2016-01-20:
**Problem:** it is hard to verify the accuracy of metadata formulas when the input values are not shown. 

**Solution::** on the /data_sources.shtm page point details tab where the metatdata script is edited:

* display the value and timestamp (in time and as an integer) beside each Point Name in the Script Context point list that is above the script edit box, 
* update the context point values each time the update event is triggered for that script, or each time any of the points is updated.
* also, display the metadata function return value and timestamp just below the input context points, above the script edit box rather than below the script edit box. This saves the need to scroll past a mostly blank edit box, and keeps the return value grouped next to the input (context) values.

resotek, 2016-01-20:
When the green 'validate script' button is pushed, all inputs should be updated. As before, the new return value should be displayed, but it should also include its timestamp.

resotek, 2016-01-20:
Also, the context points Var column is too short and frequently truncates variable names. It should be widened to at least 32 characters.

terrypacker, 2016-05-25:
I find that using the logging functionality while validating scripts to print the values of the points out during the run is useful and probably more fail safe then having a constantly changing point value on the page.

terrypacker, 2016-05-25:
I have added a long form style to the variable name input.

closed 2016-06-15 by Puckfist