MangoAutomation/ma-core-public #618: Modbus Publisher - Errors sneaking to standard out

Puckfist, 2016-01-22:
ModbusIpSlave probably needs to "implements MessagingExceptoinHandler" and call setExceptionHandler(this) around line 46. As it sits the DefaultMessagingExceptionHandler is printing the stack trace to stderr when an IOException is received in the publisher's thread.

terrypacker, 2016-01-29:
One question is as to what to do with the exceptions?  The easiest solution is to log them, the most complex would be to raise publisher events.  The problem with raising events is that it may be difficult to determine what publisher the event is for since they all share the same slave.

Puckfist, 2016-02-02:
I have confirmed that you caught the sneaky buggers. You raise a question, so I am removing the other labels and adding question as opposed to closing.

terrypacker, 2016-05-26:
I think this should be closed.

terrypacker, 2016-06-02:
Make sure these are all logged.

closed 2016-06-07 by Puckfist