MangoAutomation/ma-core-public #619: Reports - Units fail to deserialize

Puckfist, 2016-01-24:
This is a copy of https://github.com/infiniteautomation/ma-modules-public/issues/14 but I made progress and we blast through these so it's here. I took a cue from the DataPointVO class to do the storage in the database as text rather than letting the object serialize, and it fixed the issue with Fahrenheit, and so I presume it is more robust.

So, adding... for version = 4
SerializationHelper.writeSafeUTF(out, UnitUtil.formatUcum(unit));
SerializationHelper.writeSafeUTF(out, UnitUtil.formatUcum(renderedUnit));
....
else if (ver == 4) {
        	suffix = SerializationHelper.readSafeUTF(in);
            useUnitAsSuffix = in.readBoolean();
            unit = UnitUtil.parseUcum(SerializationHelper.readSafeUTF(in));
            renderedUnit = UnitUtil.parseUcum(SerializationHelper.readSafeUTF(in));
        }

to the PlainRenderer is my solution. This would have to be done for all renderers to fix Fahrenheit display everywhere (and perhaps others I don't know).

terrypacker, 2016-02-11:
Also close infiniteautomation/ma-modules-public#14 when done.

Puckfist, 2016-02-11:
I confirmed it works for the PlainRenderer, but i do not believe you did this for the other types of text renderer, did you?

terrypacker, 2016-02-17:
Changed for all renderers that extend ConvertingRenderer.  The serialization has changed for all these classes so to test:

First create a renderer of each type with existing code, then update to new code and ensure the renderers can be read and written.

Puckfist, 2016-02-17:
Confirmed new system & upgrade, everything seems okay. The two things kicking around my mind are:

1. The range-rendered point didn't have units. I think this is something to do with the range renderer, not this unit issue. I'm not sure it should display units for statistics, either, since the forming of statistics seems to forgo fitting a value into a range.
2. Our code will probably break if javax.measure.unit.UnitFormat ever implements ucum units. I think in UnitUtil, ucumFormat and localFormat are actually references to the same object. Should they ever not be, all the calls to formatUcum will break for the units we define/alias in UnitUtil and assign to the locale format. I'm not sure that'll ever happen, but UnitUtil is a very deceptive class because they are references to the same object.

Puckfist, 2016-02-18:
Accepted as fixed.

closed 2016-02-18 by Puckfist