MangoAutomation/ma-core-public #621: User permissions can add redundant tags or leading commas

resotek, 2016-01-27:
**Problem:** when applying permissions to a filtered permissions list where the tag to be added already exists on some of the datapoints in the filtered results, the tag is appended to the end of the string as ",tagnanme" even if the tag already exists in the permission field for the respective datapoint. The result is redundant tags: i.e. the same tag may appear two or more times in the tag string.

Another artifact is that formerly empty permission tag fields end up with a leading comma. Although it looks ugly, it is **unknown whether it creates undesired side effects such as a security risk**. It should be easier and more reliable to remove dangling commas from the database than to thoroughly test for possible undesired effects.

**Solution:** Check the permissions string to see if it already has the tag to be added. If does, then don't append the tag. If the field is empty, don't append a leading comma to the tag. Provide a method to clean up databases that contain redundant tags and dangling commas.

terrypacker, 2016-02-11:
One obstacle here is that for performance these updates are done in a bulk SQL update statement as iterating over 10s of thousands of points would take a very long time.  Thus adding new logic to the bulk update will be complex.  The help of an SQL savant would be appreciated.

Duplicate permissions will eventually be removed when a user is saved as that is part of the validation process on saving a user.

resotek, 2016-03-15:
>Duplicate permissions and leading commas are eventually removed when a user is saved as that is part of the validation process on saving a user

That's good to know, but does not address duplicate tags that don't map to a username, such as operator, user, etc. Perhaps a viable alternative would be a cleanup routine. That may be needed anyways to be able to clean old configurations that pre-date the resolution of this issue.

Puckfist, 2017-03-21:
#838 Seems related

terrypacker, 2018-07-16:
This is fixed in Mango 3.4.5 which will be released very soon if not today.

closed 2018-07-16 by terrypacker