MangoAutomation/ma-core-public #622: Add Datapoint Permissions filter support for missing tags

resotek, 2016-01-27:
Add Datapoint Permissions filter support to list datapoints that do not have a particular username or tag. It is common for a user to complain that they received a permission error when attempting to read a certain data point. The simplest desired action to be taken would be to list all points that the user has no permissions to, and bulk add the permission to some or all points. This is much easier than piecewise searching for individual points to change their permission.

Example use case: list all datapoints where user "bozo" does not have read permission.

Initially, I thought the first column in the permission filter list (true/false) had this effect. Through trial and error I realized that was not the case. The permissions filter help should be better documented.

Puckfist, 2016-01-27:
I feel this will be done by negating the entire filter.

Puckfist, 2016-05-27:
Not And and Not Or as filter styles have been added and tested.

closed 2016-05-27 by Puckfist