MangoAutomation/ma-core-public #624: Core - Update many points' Template: Aw Snap! chrome unhappy.

Puckfist, 2016-01-28:
I am operating on a fairly large set of points, 20,000 or so. I attempt to update the template they share in Chrome and upon clicking 'Update Template', which normally loads a list of points the template is applied to, instead sits for about ten seconds and redirects to the 'Aw Snap'

I am able to use Chrome when the list is smaller. Firefox worked on the 20,000 points system.

terrypacker, 2016-02-25:
When fixing this we should:

- pop up a modal and show a spinner in it when update template is clicked
- fix the table to be lazily loaded via a dstore
- show a spinner from the modal and don't close it until done updating

terrypacker, 2016-02-25:
Also should refresh the all points table when done, to show the new settings in the table.

terrypacker, 2016-10-28:
Added DGrid, tested with several points but haven't had time to load up a massive config and fully test.

Puckfist, 2016-10-29:
Tested with 30000 points, still crashed Chrome. I had the dev tools open, what I saw was the call to find points sharing the template (TemplateDwr.findPointsWithTemplate()) return after about 16s, then Chrome 'Ah Snapped' about 4 seconds later and it forced closed the developer tools.

Also, with such a long query time, it's probably important to give some visual cue on the update button that it has made the request, otherwise it is very tempting to press the button more than once.

terrypacker, 2016-10-29:
Made some more changes, updated the template to ~80k points and it seems ok now.  It does take a long time to generate the AUDIT events when you save, so it's possible the DWR call will timeout.  (It did not on my and it took what seemed like a few minutes).  This is only a temporary solution since the new UI will use REST and not DWR like this anyway.

Puckfist, 2016-10-30:
I like it! Very snappy, very responsive.

The very first time I tried to use it, it bounced me back to the data source's edit area as though I had just clicked the pencil to edit that data source immediately after I hit the 'update template' button to open the dialogue and the dialogue appeared briefly. Multiple attempts to reproduce afterward were unsuccessful, and the dialogue seemed to work great.

Puckfist, 2016-10-30:
When using the browser on only half my screen, the throbber for indicating the updating is in progress is obscured by the other fields/buttons in the template section of the edit point page.

Puckfist, 2016-10-30:
It seemed to handle the case of being updated by two users at the same time okay (well, one, then seconds later starting the other). The second save prevailed, but I could see the tasks were busy competing with each other to update stuff: 

DEBUG 2016-10-30 08:39:34,118 (com.serotonin.m2m2.rt.EventManager.raiseEvent:190) - Event raised: type=AuditEventType(auditType=DATA_POINT, referenceId=31898), message=User "admin (1)" changed Data point with id 31898: <br/>&nbsp;&nbsp;&nbsp;&nbsp;Logging type: "All data" to "Interval" 
DEBUG 2016-10-30 08:39:34,119 (com.serotonin.m2m2.rt.EventManager.raiseEvent:190) - Event raised: type=AuditEventType(auditType=DATA_POINT, referenceId=28828), message=User "admin (1)" changed Data point with id 28828: <br/>&nbsp;&nbsp;&nbsp;&nbsp;Logging type: "Interval" to "When point value changes" 
DEBUG 2016-10-30 08:39:34,120 (com.serotonin.m2m2.rt.EventManager.raiseEvent:190) - Event raised: type=AuditEventType(auditType=DATA_POINT, referenceId=31899), message=User "admin (1)" changed Data point with id 31899: <br/>&nbsp;&nbsp;&nbsp;&nbsp;Logging type: "All data" to "Interval" 
DEBUG 2016-10-30 08:39:34,120 (com.serotonin.m2m2.rt.EventManager.raiseEvent:190) - Event raised: type=AuditEventType(auditType=DATA_POINT, referenceId=28829), message=User "admin (1)" changed Data point with id 28829: <br/>&nbsp;&nbsp;&nbsp;&nbsp;Logging type: "Interval" to "When point value changes" 
DEBUG 2016-10-30 08:39:34,122 (com.serotonin.m2m2.rt.EventManager.raiseEvent:190) - Event raised: type=AuditEventType(auditType=DATA_POINT, referenceId=31900), message=User "admin (1)" changed Data point with id 31900: <br/>&nbsp;&nbsp;&nbsp;&nbsp;Logging type: "All data" to "Interval" 

I would wonder if there was more going on in my system (all data sources are disabled) if we think it would be possible for their ordering to switch around and result in points in the runtime getting updated such that some running points displayed the property of each template.

I don't know how concerned we are with that case.

terrypacker, 2016-10-30:
Agreed, this process should be limited to a FIFO type queue ordering based on some set of rules.  But for now I think it will be acceptable.

Thinking about how this should work in a multi-user system can be complex.  I'm moving this to 2.8.x so we can decide on a work-flow and do a more robust implementation in the REST api.  Some things to consider:

1.  Concurrent updates to different templates can happen if they don't effect the same points.
2.  Concurrent updates to the same template should not be possible.
3.  Concurrent updates to different templates that edit the same point should not be possible.
4.  Limiting all requests to a FIFO ordering is not ideal but would be by far the easiest solution.

terrypacker, 2017-04-11:
Closing and will re-open new issue.

closed 2017-04-11 by terrypacker