MangoAutomation/ma-core-public #626: Graphical Views - Non-admin Editing & Permissions

Puckfist, 2016-01-29:
If you are editing a graphical view as a non-admin, you cannot save that graphical view if there are any user groups in the permissions fields you don't belong to, even if they were there to begin with.

Puckfist, 2016-02-11:
I have confirmed that you can save with the existing permissions. I am marking it a question because you can still delete the permissions for groups you don't belong to and save the item without the old groups permissions. Is that what we want?

terrypacker, 2016-02-11:
No that is not what we want, I'll fix that to handle that case too.

Puckfist, 2016-02-11:
Got the word that you shouldn't be able to remove a permissions group from an item if you do not have that permissions group.

Puckfist, 2016-02-11:
Confirmed. One could nitpick whitespace (groups are not trimmed when checked here) issues, but it is well enough behaved.

Puckfist, 2016-02-12:
Passing tests, well behaved. One thing I'm wondering, if we care, is how much sense it makes to restrict a person with edit permission's ability to modify read and set permissions for user groups but not for the anonymous user. Any user with edit permission could modify the anonymous read and set permission, and thereby extend or revoke permissions to some set of users they may not share groups with. Do we care?

Puckfist, 2016-02-16:
The ruling came that we care, but currently only so far as including notes about things like this in an eventual security advisory to system admins.

closed 2016-02-16 by Puckfist