MangoAutomation/ma-core-public #629: Core - userEvents not purging?

Puckfist, 2016-02-01:
We should confirm the userEvents table is actually purging during the nightly purge. My sense is that it is not.

terrypacker, 2016-02-09:
I've just tested it on H2 and found that the on delete cascade is in fact working.  What database type did you have issues with?

Puckfist, 2016-02-09:
I tested it an confirmed it working on MySQL as well, and confirmed the code has been there since Mango's inception. I am led to believe someone manipulated the database in question, as evidenced in the "show create table userEvents" statements:

failing system:
CREATE TABLE `userEvents` (
  `eventId` int(11) NOT NULL,
  `userId` int(11) NOT NULL,
  `silenced` char(1) NOT NULL,
  PRIMARY KEY (`eventId`,`userId`),
  KEY `performance1` (`userId`,`silenced`)
) ENGINE=InnoDB DEFAULT CHARSET=utf8 

passing system:
CREATE TABLE `userEvents` (
  `eventId` int(11) NOT NULL,
  `userId` int(11) NOT NULL,
  `silenced` char(1) NOT NULL,
  PRIMARY KEY (`eventId`,`userId`),
  KEY `performance1` (`userId`,`silenced`),
  CONSTRAINT `userEventsFk1` FOREIGN KEY (`eventId`) REFERENCES `events` (`id`) ON DELETE CASCADE,
  CONSTRAINT `userEventsFk2` FOREIGN KEY (`userId`) REFERENCES `users` (`id`)
) ENGINE=InnoDB DEFAULT CHARSET=utf8 

Sorry I didn't check the guilty system first for any shenanigans. Closing...

closed 2016-02-09 by Puckfist