MangoAutomation/ma-core-public #631: MangoNoSql Migration - Wrong connection string, bad handling

Puckfist, 2016-02-03:
When the connection string to the database is wrong (I was running on H2, migrating H2) I got errors about not finding the table DATAPOINTS

What's worse, now I have an empty database at the path I specified that was wrong!

terrypacker, 2016-02-11:
H2 by default will create an empty database at the path you choose.  You need to put ;IFEXISTS=TRUE on the end of your string to fail if it doesn't.  I'm putting a link to the H2 database connection options page in the help file.

Puckfist, 2016-02-11:
Does it ever make sense to create a blank database and then try to move the point values table out of it? Would it be difficult to add such a parameter only to H2 connection strings for the migration?

terrypacker, 2016-02-11:
I guess we could check to see if its already in the string coming from the UI and if not add it, but I'm not sure its a big enough problem.

Puckfist, 2017-08-31:
Line 131 MangoNoSqlMigrationManager:

if("org.h2.Driver".equals(driverClassname) && !connectionUrl.contains(";IFEXISTS=TRUE"))
	                connectionUrl = connectionUrl + ";IFEXISTS=TRUE";

or we close the issue?

Puckfist, 2018-04-17:
Opted for do it