MangoAutomation/ma-core-public #632: Metadata error message "Point disabled or does not exist" does not cite points

resotek, 2016-02-04:
**Problem:** the error message "Point disabled or does not exist" does not list the name nor xid of the point that is disabled, nor of the metadata point that failed while trying to use it. This is not sufficient information to directly identify the problematic points.

>ERROR (com.serotonin.m2m2.meta.JavaScriptPointLocatorRT.createContext:70) - Point disabled or does not exist 
com.serotonin.m2m2.rt.script.DataPointStateException 
        at com.serotonin.m2m2.rt.script.ScriptExecutor.convertScriptContext(ScriptExecutor.java:64)
        at com.serotonin.m2m2.meta.JavaScriptPointLocatorRT.createContext(JavaScriptPointLocatorRT.java:67)
        at com.serotonin.m2m2.meta.MetaPointLocatorRT.pointTerminated(MetaPointLocatorRT.java:213)
        at com.serotonin.m2m2.rt.dataImage.DataPointEventMulticaster.pointTerminated(DataPointEventMulticaster.java:109)
        at com.serotonin.m2m2.rt.RuntimeManager.stopDataPoint(RuntimeManager.java:444)

 Without this information, users with many metadata points cannot easily or practically rectify the problem. 

**Solution:** add the disabled point names or xids to the error message, and add the xid or name of the metadata point that failed:

ERROR (com.serotonin.m2m2.meta.JavaScriptPointLocatorRT.createContext:70) - Point **p1234** is disabled or does not exist **in metadata point (xid=DP_MyMetaDataPoint)**
com.serotonin.m2m2.rt.script.DataPointStateException 

Issue #615 would also help avoid this problem, even though in my experience it crops up after a metadata point was working.

terrypacker, 2016-02-10:
This will have to wait until the 2.8.0 release because it will require a Core change.  The idea will be to create a new type of Exception that is thrown from the ScriptExecutor that contains both the data point ID and variable name (ScriptContextVariable).  This exception should then be caught and a message logged/event raised with the Meta point's XID and the variable name and if the data point is actually disabled or really doesn't exist.

Puckfist, 2016-05-27:
Perhaps 'missing' is too vague? You have made the handling more specific, so why not make the message more specific too: "missing from database", "no point exists with ID: {1}" or what have you.

terrypacker, 2016-05-27:
Changing message to: Point with id {1} for variable named ''{0}'' is missing from database

closed 2016-05-31 by Puckfist

terrypacker, 2016-10-31:
The commit for this also has the env property:

`runtime.javascript.metrics`

to allow timing the scripts in the log.  See env.properties for more info.